Cleric-K/FlySkyRxFirmwareRssiMod #1: ia6c firmware

jotaemesousa, 2018-07-08:
Hi! I saw your modification for the X6B and ia6b receiver and it was an extraordinary work.
I have also ia6c receivers on my mini quads and I would like to contribute to this project with the rssi modification. Can you help me developing the solution?

Thanks for your time, it is realy useful.

Cleric-K, 2018-07-10:
Hi,
I guess the firmware is somewhat similar to the x6b and ia6b, so I hope it won't be too hard.
I don't have such a rx so the questions are:
* what mcu it uses
* does it expose the swclk and swdio pins to pcb pads

If you are able to dump the existing firmware with stlink I can take a look.

jotaemesousa, 2018-07-10:
Hi! Yes the receiver uses a STM32F03 (like you can see here http://rc-aviation.ru/images/obzor/apparatura/TGY-iA6C-3.jpg).
The board has the SWD pins exposed, so it won't be a problem to flash. 
I has the receiver mounted on my quad, so I will send you a dump of a firware that exists (https://github.com/povlhp/FlySkyRxFirmware/blob/master/firmwares/iA6C.bin).

I program these type of microcontrollers at work, but I am not an assembly expert, so if you need so help I can test the firmware or help you to find the solution. Moreover, this receiver has a method of Auto-bind, so maybe it will be a little diferenf from the x6b.

Cleric-K, 2018-07-11:
Hi, I looked at the binary you sent me and it is quite similar to the other ones, so making the patch was relatively easy. I don't have a way to test it so I really hope I didn't mess anything up.

Check out [IA6C_rssi.zip](https://github.com/Cleric-K/FlySkyRxFirmwareRssiMod/files/2184163/IA6C_rssi.zip).

!!! Don't forget Backup your current firmware !!!

If it turns out to be working I'll include this in the project.

jotaemesousa, 2018-07-11:
Hi, good news! It works on ia6c, I only tested on betaflight, but I think it is ok (when flying).
Thanks a lot.

Cleric-K, 2018-07-12:
Great! I'm glad it worked out!
I've pushed the source code.

Enjoy!

closed 2018-08-13 by Cleric-K