Cleric-K/FlySkyRxFirmwareRssiMod #2: x6b failsafe issue

vvac, 2018-08-23:
Hi,
    the i6x version of RSSI modified firmware appears to be based on the original firmware that had an issue with failsafe when using ibus in BF/iNav (holds channel values by default). Could you please source it with from the fixed one? It is available here: 

https://github.com/povlhp/FlySkyRxFirmware/tree/master/firmwares. 

Best Regards,
Tomasz.

Cleric-K, 2018-08-23:
Thank you for your feedback. Can you tell which model receiver you have in mind? IA6B?

vvac, 2018-08-23:
Sorry, I meant "x6b" not "i6x".

Cleric-K, 2018-08-24:
I binary-compared the the x6b firmware [that I use](https://github.com/Cleric-K/FlySkyRxFirmwareRssiMod/blob/master/src/X6B/X6B_orig.bin) and [the one](https://github.com/povlhp/FlySkyRxFirmware/blob/master/firmwares/X6B.bin) at povlhp repo - they are the same (except for the tx id bytes, which is normal).

Are you sure you have ever witnessed a x6b receiver which does not have the problem you are referring? If you have such a receiver maybe you can dump its firmware and I'll compare them again?

vvac, 2018-08-28:
I'm pretty sure that I had my failsafe working (without setting it in transmitter) before upgrade to RSSI mod.  Sadly didn't perform a firmware backup :(  Some of my friends still use unmoded versions of those receivers. I'll try to find one that works and feed you with firmware from it.

vvac, 2018-08-28:
I just realized that maybe it was the OLD version that produced the desired behavior. I mean all I know is that it had broken failsafe on iBus but maybe "broken" means that in case of RF link loss it generated no iBus frames at all, instead default TX settings (channels hold) making it possible to FC act properly in this situation. This kind of behavior is safer and more intuitive in my opinion but still have to be considered "broken" since it sends no frames when you set failsafe on your TX to something else than default. If it'd be the case it's still worth to have the older version of firmware RSSI modded as it is more convenient for use with FC featured UAVs. What do you think? In any case I'll let you know what i found.

vvac, 2018-08-28:
Does your FC failsafe trigger when you have your failsafe at default (all channels off) on tx and you turn off your transmitter? That is the scenario I can not achieve on RSSI version of firmware and worked on my previous firmware. You didn't need to set the failsafe in TX. Disconnecting RF link automatically triggered failsafe on FC.

Cleric-K, 2018-08-28:
Tonight, when I get home, I'll do more extended tests and I'll tell you.

I have never tried to make the rx _stop_ outputting frames when signal is lost. Making the rx output invalid channel value on sig loss seems to work fine.

But I'll do some tests just in case.

vvac, 2018-08-28:
Of course it works fine but (correct me please if I'm wrong) still to make it work you need to:
a) move some channels endpoint to the value that allows you to have < 950 us 
b) configure your TX failsafe 
c) move the endpoint back to minimal 1000us

This other approach I described just works out of the box and can save lots of newbies' UAVs ;)

Cleric-K, 2018-08-28:
Well, yeah.
Ok, so if you find the old firmware it won't be hard to patch it.

The question is if there are other things changed in the new firmware which might make it better in other directions. Because in order to avoid few clicks on the tx you are giving up a potentially improved firmware :)

vvac, 2018-08-28:
Yeah, maybe - I don't know really. My point was just to change this default behavior to trigger FC failsafe. But like you said, I'm also not so sure if it's worth giving up a potentially improved firmware. The sweet spot would be to hack this new firmware in such way that it sends invalid iBus frames (or stop sending them at all) but only when your tx's failsafe is all off, you think it's even feasible? ;-)

Cleric-K, 2018-08-28:
It surely _is_ possible, but I don't know if I'll have the time to do it anytime soon. I'm not a specialist in ARM disassembly - actually I did this whole project because no one has done it and I really wanted to have this feature - so I had to do it myself :) Prior to that I have never looked into ARM code, or reverse engineered anything. But thanks to several very important guidelines given to me by qba667 and several sleepless nights I managed to track down the the code which generates the IBUS frames and to inject the patch there. I guess it will take me just as many to figure out how the failsafe handling works :)

closed 2019-03-26 by Cleric-K