Cleric-K/FlySkyRxFirmwareRssiMod #6: A8s sbus not working

laurentopia, 2019-07-16:
When I set sbus in betaflight and sbus in the evo, moving the sticks don't show in betaflight

laurentopia, 2019-07-16:
[image]

Cleric-K, 2019-07-16:
As we commented with @fxbisto in #5 it seems that A8S and 8A receivers don't care about SBUS/IBUS settings made on the transmitter side. Switching between the two is accomplished at the receiver by holding down the _bind_ button for 2 seconds.

fxbisto, 2019-07-16:
Funny that this should come up twice in 2 days all of a sudden! @laurentopia how come you are using SBUS over iBus? It's actually a bit slower :)

laurentopia, 2019-07-16:
> As we commented with @fxbisto in #5 it seems that A8S and 8A receivers don't care about SBUS/IBUS settings made on the transmitter side. Switching between the two is accomplished at the receiver by holding down the _bind_ button for 2 seconds.

ohhhh I didn't know that, thank you.

@fxbisto because I wanted to also try the rssi_src_frame_errors=ON so i hooked my a8s to the sbus, ibus may or may not support packet drop. What do you guys think?

fxbisto, 2019-07-16:
I'm not sure that it would support it, flysky's SBUS implementation is fairly basic from what I've read. The native RSSI on iBus works really well, you won't need more than that I promise! Just flash the 8 or 14ch modified firmware from here, 14ch is fine for bF. Disable the RSSI_ADC option in bF and then set RSSI aux channel to 10 and you're all set.

Cleric-K, 2019-07-17:
@fxbisto is correct.
First, these receivers do not detect dropped frames - they send ibus/sbus frame only when an air frame is received from the transmitter. As a result if nothing is received from the transmitter, nothing is sent to the fc (that is, the rx does not even try to detect if an air frame should have been received but it did not come).

Second, and more importantly, `rssi_src_frame_errors` is a only useful if you _don't_ have any other means for measuring rssi. It is mutually exclusive with normal rssi sources. It is kind of a "still better than nothing" option, since it does not really measure signal strength, but only gives indication when signal is already worse enough that frames are dropped. It is explained in the docs:
>Note that RSSI stands for Received Signal Strength Indicator; the detection of S.Bus dropped frames is really a signal quality, not strength indication. Consequently you may experience a more rapid drop in reported RSSI at the extremes of range when using this facility than when using RSSI reporting signal strength.

closed 2019-10-25 by Cleric-K