Cleric-K/FlySkyRxFirmwareRssiMod #7: X6B read locked - and I messed up

CoreyOsgood, 2019-10-23:
I connected a new FS-X6B (bought about 2 weeks ago from Hobby King) to my ST-LINK V2 (updated to latest firmware), attempted to read it, and got the message

> Can not read memory!
> Disable Read Out Protection and retry.

So, I opened up the Option Bytes menu and set Read Out Protection to Level 0. That was my mistake. Apparently this completely erases the device memory, and it's now bricked. The LED no longer turns on, with or without the Update button pressed, and it never responds to the ST-LINK. Any hints for unbricking it?

Cleric-K, 2019-10-23:
Hi, you should try `Connect Under Reset`
[image]

[image]

Locate the RST pad. ST-Link does have a RST line, but I was never able to use it. Instead you have to do it _manually_.

1. Connect the `data`, `clock` and `gnd` lines as usual.
2. You need to supply 0V (`gnd`) to the reset pad, but in such a way that you can easily detach it. While 0V is applied to that pad the MCU is kept at reset state but it still reacts to the SW signals.
3. Set up the the `Connect Under Reset` and press `Connect`.
4. Initially nothing will happen but then release the 0V at the `RST` and ST-Link should connect

CoreyOsgood, 2019-10-23:
Thanks so much! It lives again!

Also, PSA, DO NOT USE TWISTED PAIR WIRES. I think that was the root of my problem, the wires I pulled out of my scrap drawer to use for data and clock were twisted together. Even with manual reset, I couldn't connect, until I realized my mistake and untwisted them. As soon as I did, solid connection first try, reprogrammed, set 0x1c00 to a random hex number, and bound to my receiver.

closed 2019-10-23 by CoreyOsgood

Cleric-K, 2019-10-24:
Great news!

Thanks for the tip about the twisted wires. I find it little strange that such small currents are able to interfere, but important thing is that it worked for you.

Bucc1o, 2020-10-23:
Hi all i have the same problem i have done what Cleric wrote, but i'm not able to connect.. It seems ti connecting but after the program say unable to connect.. what can i do?

Cleric-K, 2020-10-23:
It should work.
1. Close the program
2. Unpluging the STLink
3. Plug it again
4. Start the program
5. Go to settings and change to Connect Under Reset
6. Hold GND connected to the RST pad
7. Click Connect in the program
8. Release RST

Try several times if it does not work. The MCU can not be bricked easily.

Bucc1o, 2020-10-23:
Sorry i have opened also a new issue, because this is closed.. anyway i tried several time and get the same results, i don't know why.. maybe a tracer log can help?

elsp1991, 2021-12-21:
Hello Guys,

I also have a X6B that I wanted to flash it using a J-link tool but got this message that "Device is lock and unlocking it will cause mass erase" does this mean that none of us here has a backup of the original firmware for at least the X6B?

It took ages this thing to arrive from china I dont wanna kill it before even see it working :).

Cleric-K, 2021-12-21:
> Hello Guys,
>
> I also have a X6B that I wanted to flash it using a J-link tool but got this message that "Device is lock and unlocking it will cause mass erase" does this mean that none of us here has [...]

You can get the original fw here https://github.com/Cleric-K/FlySkyRxFirmwareRssiMod/blob/master/src/X6B/X6B_orig.bin
Not all rxs are locked. At least this used to be the case. Maybe newly produced are.

cohibamatt, 2021-12-29:
Thanks Cleric,
I also am having problems. I couldn't get st-link utility to work so I downloaded the cube programmer. When I tried to flash the firmware it said it was locked and wouldn't start the core, it was bricked. I was able to get back into it by following the connect under reset but I still can't flash. is there much difference between the two programs?  This is kinda new to me. I ditched windows for Linux and there is a steep learning curve for me.