Cleric-K/FlySkyRxFirmwareRssiMod #8: could you upload original firmware?

viktortrass, 2019-10-24:
Hello, I had a problem with my x6b receiver, first I tried to connect to target from st-link utility but cannot read memory because it has read out protection I don't know why(I think what I have read on internet could be when I open st-utility first time and set size to 0x8000 instead 0x1000 this change lock the memory but I'm not sure) , second I try bytes option to set level 1 to 0 but I didn't see that this mass erased the chip before RDP is set to 0 and I hadn't any firmware after that and now I couldn't connect to the chip in normal or under reset mode so finally I connected to the chip putting in hot plug mode(only mode that worked for me I read on internet too) and flashing your modded firmware for channel 14 but now I hadn't a backup of my original firmware, it would be perfect if you could upload it thought I think I will never use it again. Thank you.

Cleric-K, 2019-10-24:
Hi. The firmwares are available at this repo: https://github.com/povlhp/FlySkyRxFirmware

Can you, please, explain about this hotplug mode? In situations like this I use Connect Under Reset and it works for me.

viktortrass, 2019-10-24:
I hadn't used the receiver before, it arrived I soldered the pins to st-link v2 and installed stutility and driver, I only changed size to 0x8000 and connect to target but it told me that cannot read memory, is locked so I entered bytes option and change to Level 0 as I told you before but this mass erase all before the readout memory protection goes out and now the stlink utility doesn't recognize and red led doesn't turn on(there wasn't firmware inside a chip), I couldn't connect in under reset mode I tried with the RST PIN connected too but still didn't work, I followed reading on internet and find a person who only worked on hotplug mode so I selected it and suddenly it read the chip, after that I flashed your modded firmware and put back in normal mode and all worked again fine.

I read too that some cheap stlink v2 programmer can have pin of SWCLK and SWDIO changed but I don't know really, I think mine was not changed because I wouldn't could program it if this are changed.

closed 2019-10-25 by Cleric-K