Cleric-K/FlySkyRxFirmwareRssiMod #9: New Receiver FS-X14S

philip1987, 2019-12-04:
First off, thank you for your excellent work! I already installed the mod on a FS-A8S.
FlySky has recently launched a new receiver that supposedly extends the range to over 1.5km. Which is very exciting if true. I haven't tested it yet but here is the manufacturers own test: https://www.youtube.com/watch?v=TCdAkFM6Va4&t=23s

My question: I'd like adapt your RSSI mod to the FS-X14S (which would be very useful considering the extend range of the receiver). Could you point me in the right direction of how to do so? (I imagine the first step is to extract the firmware.) If it's successful you can add the mod to your list of modded receivers so others can also benefit from it.

MicroMontis, 2019-12-05:
Thanks to the Clerik, it has been just patched! Wait for the patched firmware to appear in this repository or just download it from here: https://www.rcgroups.com/forums/showthread.php?3093552-RSSI-Firmware-mod-for-FlySky-IA6B-and-X6B/page23

philip1987, 2019-12-05:
Thanks MicroMontis for pointing me to the thread on rcgroups! I'll try it out asap. Just a quick question since you already flashed it: did you attach SWDIO to the pad labelled with "D", and SWCLK to the "C" pad? This thing is really tiny! I'm excited to test out the range (which is less scary with RSSI on the OSD).

MicroMontis, 2019-12-06:
Yes, the tiny solder pads near the antenna are 3.3V, GND and C goes to
SWCLK, D goes to SWDIO.

Good luck!

closed 2020-02-02 by Cleric-K