Cleric-K/FlySkyRxFirmwareRssiMod #10: Conntected error IA6B

tn2009, 2020-01-25:
Hi. I connected IA6B as it should be. I was first shown in the ST Link utility program that there is no access to memory. I'm in the Target - Option bytes menu... chose the Read out protection Level 0. After saving the standard firmware from A6B to your computer. After clicking Disconnected from the target. Then Connected to the target... And no longer connects... The program writes this

22:46:54 : Can not connect to target!
                  If you're trying to connect to an STM32W1xx device, please select Normal or HotPlug mode from Target->Settings menu.
                  If you're trying to connect to a low frequency application , please select a lower SWD Frequency mode from Target->Settings menu.
22:46:55 : Can't reset the core

Please tell me what to do?

Cleric-K, 2020-01-25:
Changing the Readout protection erases the flash I think. That is, there's no point in the ROP if you can just flip the bit and read it out. So it allows to the remove the protection but it also erases the flash memory.

You can try Connect Under Reset.
Locate the Reset pad near the Data and Clock.
You'll need some way to supply 0V (GND) to the RST pad by hand (so that you can remove it at anytime).
1. Restart both STLink Util and replug the stlink usb just to make sure you are starting clean.
2. Open the settings in Stlink util and change connection mode to Connect Under Reset. Do not connect yet.
3. Hold 0V at the RST pad. While you are holding it you are keeping the MCU in reset state (but still it allows some special communication over the debug pads).
4. Press connect in stlink util.
5. Initially nothing happens, then release the GND from the RST pad. It should connect.
6. Flash the firmware normally.

tn2009, 2020-01-25:
Recovered. Everything works. Thanks! Why is RSSI always 98 percent instead of 99. Is this how it should be? INav 2.3 firmware

Cleric-K, 2020-01-25:
What is the channel value when looking in the receiver tab of the configurator (in the 1000-2000 range)?

tn2009, 2020-01-26:
1990

Cleric-K, 2020-01-27:
Well, +/- few percent is in about the margin or error, so I wouldn't be concerned.

closed 2020-02-02 by Cleric-K