Cleric-K/FlySkyRxFirmwareRssiMod #11: Flysky X14s

Dropsleutel, 2020-02-24:
I flashed the firmware on Flysky X14s,  but can someone confirm that this receiver can provide full bidirectional single wire telemetry capable Ibus ? 

I assume Transmitter must be set Ibus, at the Receiver the Sbus pin will than be outputting Ibus? 
Will it also be bidirectional capable, so accept telemetry from flight controller?

To do so I want to connect to Uart Tx1 (uninverted) betaflight:
I enable serial Rx, select ibus
I enable telemetry

in CLI set the port:
serial 1 4160 115200 57600 115200 115200
set serialrx_halfduplex = ON
set tlm_halfduplex = ON
set serialrx_inverted = OFF
set tlm_inverted = OFF

any succes stories on this?

Cleric-K, 2020-02-25:
Single wire communication is _not_ supported by any flysky receiver. It is just not part of the protocol specification.

Furthermore, from what I know, telemetry works only on the IA6B/10B receivers through the `SENS` pin.

fxbisto, 2020-02-25:
@Dropsleutel the "single" part of single wire is only on the flight controller side. You need to use the 2 dedicated pins (iBus out + iBus sensors) from the receiver with a diode+resistor network to connect them to the same UART pin on the flight controller.
See the bF wiki:
https://github.com/betaflight/betaflight/wiki/Single-wire-FlySky-(IBus)-telemetry

Dropsleutel, 2020-02-26:
Ah ok, I was quite certain that Ibus was bidirectional single wire capable like Frsky Fport uses one pin conneted to on Tx Uart at flight controller, but I now understand you have to make the "combiner".

So than another question to  X14S.  Originally according he included manual, it has Sbus out. (Sbus is inverted signal)
So when flashed X14S can it send ibus output and is it then uninverted ? (Sure to select ibus at Transmitter)

Cleric-K, 2020-02-26:
The signals are always correct polarity. The UARTs of the MCUs which all flysky receivers use, support software inversion so it is not a problem to support the inverted SBUS.

The one thing I'm not sure about is if these x14s cares about the transmitter setting. If I remember correctly a8s switches between ibus/sbus by holding the button for several seconds and the transmitter setting has no effect. Since x14s is similar in firmware to a8s it may be possible that the protocol selection works in similar way.

Dropsleutel, 2020-02-26:
Thanks about sbus/ibus,  it matches my info. 
You select at the transmitter side to stream Sbus or Ibus from the Rx output. (at least in deviation Tx it is possible) When set Ibus, you need uninverted. when Sbus you need to connect to inverted Rx.

I am doing some more testing as I am bit of confused.
I had in mind I did connected the Flit  receiver in the past, (with out the diode combiner, just the ibus output) similar like Fport, so wired to a Tx pad (uninverted) at the betafflight controller and then selected half duplex, using port settings as per Betaflight instructions, I thought that it worked. 
But I am now not so sure anymore, need testing.

Dropsleutel, 2020-02-26:
just tested on F4, with ibus iA6C receiver, I can confirm I get the Rx working on two different ways, either connected to an Rx pin, either connected toa Tx pin. I receive only signal and receiver voltage telemetry. (but those are defaults ) ia6b, it has not 2 outputs.

this works, tested IA6C,  ibus wired to Rx 1

set tlm_inverted = OFF
set tlm_halfduplex = ON
set serialrx_provider = IBUS
set serialrx_inverted = OFF
serial 0 64 115200 57600 0 115200

This works also,  tested IA6C, ibus wired to Tx 1
Telemetry output must be set to ibus in ports table, to have Rx working in betaflight.
But no need to enable telemetry feature in configuration, on or off does not matter.

set tlm_inverted = OFF
set tlm_halfduplex = ON
set serialrx_provider = IBUS
set serialrx_inverted = OFF
serial 0 4160 115200 57600 115200 115200

Now need to test ibus setup with X14s

Dropsleutel, 2020-02-26:
Hmm ok guys I am getting confused can someone confirm that the firmware posted for X14S actually works? I can confirm it binds, but are we sure the ibus/sbus output pin is sending data?

Cleric-K, 2020-02-27:
I haven't personally tested X14S because I don't have one. But it has been confirmed that it works:
https://www.rcgroups.com/forums/showpost.php?p=43311399&postcount=349

Furthermore, X14S (similar to A8S and 8A) does not have telemetry at all (by design).

Dropsleutel, 2020-02-27:
Ok I did some more testing and soldering. The X14S did does not output IBUS neither with the original firmware it works only on SBUS.  (and I am now sure I did connect right as compared with my IA6C) 
You can change setting to Ibus or Sbus in the Txransmitterm, in both cases the Rx will output Sbus.
Further I can confirm that the modded firmware does indeed provide the RSSI.

So question: Can we get Ibus output ?
Second can we get the telemetry pin, or solder small wire to CPU. I like audible Vbat alarm via Telemetry.
Do we have schematics of the flysky receivers very likely they are all similar?

Cleric-K, 2020-02-28:
It is _not_ possible to have telemetry with x14s. The code for that simply does not exist in the firmware.

Did you try switching IBUS/SBUS by holding the bind button for few seconds?

Dropsleutel, 2020-02-28:
A pity it does not work,  S-Bus only.
As all Flysky/Flit receivers it use the same chipsets, rather then some Pin IO targets, why can it not all run the same firmware, so all with telemetry ibus ?

Cleric-K, 2020-02-28:
Hm.. seems you are correct - this rx only supports PPM and SBUS according to the [manual](https://static1.squarespace.com/static/5bc852d6b9144934c40d499c/t/5db94b84d6d3d15924964d85/1572424598511/FS-X14S说明书页20190902.pdf).

About telemetry - x14s, a8s, 8a use MCU with only 16KB Flash memory which is almost completely full already (other receivers have 32KB). There's simply no space to implement telemetry.

Dropsleutel, 2020-02-28:
Ah that clarifies but anyway keep up the good work thanks!!, I will fly it on S-Bus using your RRSI code. Not sure ibuy, more I would hoped for single wire ibus with telemetry and small.

The FLIT functionally it is fine, but I do not like so much as range is poor compared to original Flysky, , I tried Freq tune but it did not help. 
For Flit no clue why range is poor, maybe LNA not properly wired or controlled? 

So still no perfect Flysky Rx....

closed 2020-04-03 by Cleric-K