Cleric-K/FlySkyRxFirmwareRssiMod #14: X14s bricekd after the update

lolzika, 2020-08-17:
Hello,
I flashed my receiver today with an stlink v2. After the flash it seems bricked. Doesnt power on at all. No led no heat of of it. It was brand new out of package. I powerd it with 3.3v gnd and connected clock and data interface so my stling. Dumb me created no backup. Is there any chance of saving the thing or is it 100% gone. When i connect it to the stlink again there is 0 signs of life. It doesnt even connect to the stlink.

Cleric-K, 2020-08-18:
Hi. It should be possible to connect the stlink using the so called "Connect Under Reset". For this to work you have to locate the `reset` pad:
[image]

Connect everything else as normal. You'll need some way to provide low level (`GND`) to the _reset_ (`R`) pad. One way is to use a jumper wire - one end at `GND`, the other holding manually touched at the `R` pad.

While `GND` is provided at `R` the microcontroller is kept at _reset_ state.

At STLink you'll have to make this change:
[image]
(in my experience the ST-Link sometimes stops working. That's why when in doubt, I always re-plug the St-link in the USB port to make sure I'm starting "clean")

Then the procedure is:
1. Hold `GND` connected at the `R` pad.
2. Click `Connect` at `ST link util`. Initially nothing will happen.
3. Release `GND` from the `R` pad - the connection should be established.

Then you can flash as usual.

lolzika, 2020-08-18:
hello. i have connected successfully but my firmware still doesn't work.
coule you maybe send me the original firmware binary or whatever I need for
flashing this.

thanks

best regards
Darije caic
fpv pilot

Cleric-K, 2020-08-18:
You can get the original fw at
https://github.com/Cleric-K/FlySkyRxFirmwareRssiMod/blob/master/src/X14S/X14S_orig.bin
`Download` button on the right.

I hope it works.
Maybe if your receiver is some new version with different hardware that could be the problem...

lolzika, 2020-08-18:
sadly still doesn't work. the LEDs don't light up and there aren't any signs of life. I had to connect with a reset every time even after re-downloading your firmware. I gave up on the receiver. thanks for your help and the great flysky receiver support. I think I might have messed up somewhere in the physical processes sadly.

Cleric-K, 2020-08-18:
That's strange. If flashing works it means that the MCU is OK. So the flashing reports success?

Can you take a look at the `Option Bytes` in st link util? (once you connect the mcu).
It should look something like:
[image]

If `WDG_SW` is checked, maybe you can try clear it and `Apply`. If possible send me a screenshot of that window.

lolzika, 2020-08-18:
[image]

Cleric-K, 2020-08-18:
Ok,
well you can try unchecking the `WDG_SW` and `Apply`. We are shooting in the dark, just trying random things. I don't know at this point what else we can try.

lolzika, 2020-08-18:
now it connects without resetting it. still no led. what is the next step? currently have original firmware flashed on it

lolzika, 2020-08-18:
I flashed your firmware now and it connects without the need for a reset.

lolzika, 2020-08-18:
reciever leds are up and working.

lolzika, 2020-08-18:
i will test now and see what is the result with your rssi firmware. if it works I will be soo thankfull

xctmaf, 2020-09-27:
> Can you take a look at the `Option Bytes` in st link util? (once you connect the mcu).

Will this option be same for all another receivers?
IA6B/IA10B/X6B/IA6C/X8B/IA8X/A8S/8A/X14S/Flit10/RX2A Pro/Pro V1

Cleric-K, 2020-10-01:
I think all have the option byte but I don't know if anyone else had similar problem and had to switch the watchdog (SW_WDG) off