Cleric-K/FlySkyRxFirmwareRssiMod #15: X6B bricked

Bucc1o, 2020-10-23:
Hi all i'm new and i need an help.. i tried to flash my x6b with stlink utility, first time i don't get it working but after switching the power form 3.3v to 5v (+vcc pad on the receiver) it worked but the utility say to me "Can not read memory! Disable Read Out Protection and retry"
.. after a research i set on Option Bytes menu, Read Out Protection to Level 0.. now my receiver is bricked i think.. the red led no longer turns on.. on another closed issue here i see another member with the same problem, so i follow the procedure wrote by Clericto connect under reset the board. but it not work properly.. (image attached).
what can i do? 
[image]

Bucc1o, 2020-10-23:
This after i press ok button 
[image]

Cleric-K, 2020-10-23:
Which of the wires you have connected between the stlink and receiver?

Bucc1o, 2020-10-23:
I have connected the 3v3, dat, clk and gnd.. the rst connected to other gnd on the st link ( i release it once clicked connect on st link utility)

Cleric-K, 2020-10-23:
It's better to power with 5V from stlink and not use 3.3V otherwise it looks as it should.

Bucc1o, 2020-10-23:
Sorry i don't understand what you said.. so i have to connect It under reset using 5v from st link, gnd, dat, clk and RST manually releasing it when connect?

Bucc1o, 2020-10-24:
I've solved the problem.. today i tester with a tester each wire and solder and found one wire broken (the gnd) so i replaced it with a new one and follow tour istruction using 5v instead of 3v and It works.. i flashed the firmware successfully.. thank you so much Cleric! 

For those who have the same problem, make sure of your wire and solder and if needed use 5v from st link instead of 3v

Cleric-K, 2020-10-25:
Great! I'm glad you figured it out.

closed 2020-10-25 by Bucc1o