Cleric-K/FlySkyRxFirmwareRssiMod #16: fs-x14s receiver not binding with fs-i6x TX

Georgesorin, 2021-06-25:
Hi, I just received my new flysky fs-x14s receiver. But i cannot bind it with my transmitter (fs-i6X). The green led from the receiver is flashing and on the tx screen is remaining on "RX binding ..".
I have tried to update my tx and also my rx but no success binding.
Also I have tried:
*Setup the TX on SBUS => still not binding
*Factory rest TX => still not binding...

Do any of you encounter this problem? any suggestion will be much appreciated. 
Thank you in advance.

Cleric-K, 2021-06-25:
Are you sure the RX is not bound? For receivers without telemetry (such as fs-x14s) it's not possible for the TX to know that binding is complete because the RX doesn't send anything back. Can you verify through betaflight if the receiver is working?

Georgesorin, 2021-06-25:
I just tried your suggestion. I have configure betaflight to read the receiver. 
Already setup the TX on binding mod and turn on the receiverwith the bind button press (is start flashing green)
afte 5=10 seconds (the Green led is still flashing) i have turn off everything and turn back on to see in betaflight if i have any commands received from my TX. Nothing at all, and also the receiver slow flashing the red led.

Cleric-K, 2021-06-25:
Strange. I don't really have any idea what that could be. Was the RX working before you flashed it? Can you try flashing it again with the original fw?

Georgesorin, 2021-06-25:
Nope....the RX doesn't want to bind even with the stock firmware this is why i have update with this firmware maybe it will work this way. But it doesn't :(

Cleric-K, 2021-06-25:
Unfortunately, most probably the RX is faulty.

Georgesorin, 2021-06-25:
I was afraid of this but...meh, what can i do...

Thank you so much for your time.

closed 2021-06-25 by Georgesorin