Cleric-K/FlySkyRxFirmwareRssiMod #17: Possible to flash this Receiver?

benb0jangles, 2021-09-03:
Hi, I have been given a receiver from a 'flying Fy-X8' quadcopter. The quadcopter looks like this one:

[](https://www.google.com/search?q=dlying+fy+x8&tbm=isch&ved=2ahUKEwibkICHp-LyAhVKXxoKHSatDtYQ2-cCegQIABAA&oq=dlying+fy+x8&gs_lcp=CgNpbWcQA1C7DFjSFmCBGWgAcAB4AIABQIgBnwOSAQE3mAEAoAEBqgELZ3dzLXdpei1pbWfAAQE&sclient=img&ei=49ExYduSCMq-aabaurAN&bih=940&biw=1511&client=firefox-b-d)

I am thinking it is similar to Flysky Ia10b receiver?

FY-X8 receiver:
[image]
[image]
[image]
[image]

Flysky ia10b receiver:
[image]
[image]

The FY-X8 receiver has no pin connection for ibus servo/sens.
However, looking at the pcb picture there are 2 extra 3-pin pcb pads - which might suggest they are the 'missing' ibus servo/sens connections?

Do you think it is possible to flash it with ia10b firmware? I have no ia10b to compare pcb traces, just internet photos.

Cleric-K, 2021-09-03:
Well, if the current firmware is not locked you can back it up and try the ia10b. There's always slight risk of damage if the hardware is different and some pins that should be inputs become outputs and put voltage in the wrong places. But if you're willing to take the risk you can try.

benb0jangles, 2021-09-04:
assist me with SWD pinout?
[image]

Cleric-K, 2021-09-05:
[image]

These are the pins according to the datasheet. You can see if they go to the pads.

benb0jangles, 2021-09-07:
unable to connect
[image]

benb0jangles, 2021-09-07:
oh cool were in. I had to change to software system reset from hardware system reset in settings.
[image]

benb0jangles, 2021-09-07:
here's the original firmware:
[FY_Q7_RX_Original_Firmware.zip](https://github.com/Cleric-K/FlySkyRxFirmwareRssiMod/files/7123461/FY_Q7_RX_Original_Firmware.zip)

benb0jangles, 2021-09-07:
do you thik it can be changed to pair with the flysky i6 somehow?

Cleric-K, 2021-09-08:
> do you thik it can be changed to pair with the flysky i6 somehow? I probably shouldn't try flashing with ia10b firmware until the original bin file has been looked at.

What do you mean? That this receiver currently can't bind to i6? Then what is it compatible with?

benb0jangles, 2021-09-08:
I think it only binds to the flying fy x8 transmitter, it looks like a flysky i6
[image]

Cleric-K, 2021-09-09:
If this receiver doesn't bind to i6 chances are they use some modified flysky air protocol. Theoretically there should be no problem to use the hardware base also for regular flysky (since the RF chip is the same) but that could mean writing new firmware.

Another thing. Does the current firmware have telemetry? Does the above tx show some signal level?

benb0jangles, 2021-09-09:
Yes it has telemetry, the receiver directly connects to tx and rx on an APM board. There seems to be two versions of FY-Q7-RX receiver, one with six signal pins:
[image]

and one that I have with more pins similar to the ia10b:
[image]

interestingly, my receiver with more pins, it has a sticker over the old sticker like this:
[image]

The Flying X8 transmitter displays APM telemetry on screen:
[image]

I think qba667 made MavlinkToibus to allow APM telemetry through an ia6b receiver using an arduino pro micro between ia6b and APM:
https://www.rcgroups.com/forums/showthread.php?3020949-FlyPlus-firmware-for-FlySky-I6X
https://github.com/qba667/MAVLinkToIbus

Somebody here attempted to flash the transmitter with 10ch mod firmware and succeeded, but I am unsure if they backed up the original TX firmware firstly:
https://translate.google.com/translate?hl=en&sl=auto&tl=en&u=https%3A%2F%2Fladecadence.net%2Fblog%2Fflasheando-una-flying3d-fy-q7%2F

Somebody on rcgroups seems to think the receiver is also similar to fs-ia10b:
[image]

benb0jangles, 2021-09-09:
Also seems accroding to rcgroups comment that binding plug of receiver goes not to pin B/Vcc like normal receivers, but to port marked Tx:
[image]

So I shall try this also.

benb0jangles, 2021-09-09:
more internet searches and it seems there are also 2 versions of flysky ia10 receiver, 1)ia10 and 2)ia10b:

ia10:
[image]

ia10b:
[image]

Seems the FY-Q7-RX is more similar to ia10 receiver. 

I have not seen ia10 before, and have not found any images of the internal pcb yet.

benb0jangles, 2021-09-09:
more data on ia10 receiver:
[image]
[image]
[image]
[image]

benb0jangles, 2021-09-09:
I managed to find photos of fs-ia10 receiver pcb:

[image]
[image]

benb0jangles, 2021-09-09:
FY-Q7-RX and Flysky ia10 receiver have identical pcb:
[image]

benb0jangles, 2021-09-09:
If anybody has ia10 receiver and can make fimware copy?

benb0jangles, 2021-09-09:
Nevermind I think I found the ia10.bin file on the internet:
https://www.dropbox.com/s/fyohbyzr3rievf5/iA10.bin?dl=0

I will try to flash it to my funky receiver soon.

benb0jangles, 2021-09-09:
Neat. So I can flash original ia10.bin firmware to FY-Q7-RX receiver:
[image]

Solid LED after binding to Flysky i6 transmitter:
[image]

PPM 8 channels on receiver pin 1 working.

I guess it is now a Flysky FS-ia10 for no money.

I will test ibus also but I have to solder dupont connector to the pcb pads first.

Also, after I have done this I will try and flash FY-Q7-RX_original_Firmware.bin back to see if works.

I would like to somehow find a Flying X8 FY-Q7 transmitter firmware and flash that to the Flysky i6 Transmitter, then connect to my pixhawk and see what telemetry comes to the transmitter screen.

Cleric-K, 2021-09-09:
Great! To be honest I didn't expect this to work :) The receivers seemed too dissimilar.

If you're feeling adventurous you can play with TX firmwares through ST-Link. These TXes use STM32 mcus (of the larger kinds with many pins). I haven't opened i6 but i6x has pads for SWDIO and SWCLK. You can dump the fw from Q7 and try to flash it to i6 (after backing it up of course).

Yet always keep in mind that if there are hardware differences this can result in malfunctions (if mcu pumps voltage at pins which are otherwise supposed to be inputs).

benb0jangles, 2021-09-09:
I am relying on someone else out there to upload the transmitter firmware for me to test as I don't have it. Nice to meet you Cleric, I will comment again if the ia10.bin firmware has rssi so you can post it.

benb0jangles, 2021-09-09:
Here is the ia10.bin firmware with RSSI working:
[iA10.bin.zip](https://github.com/Cleric-K/FlySkyRxFirmwareRssiMod/files/7137908/iA10.bin.zip)

benb0jangles, 2021-09-09:
14ch ibus working too. 
[image]

benb0jangles, 2021-09-13:
So, after looking at the binary file I noticed a difference in the FY-Q7 firmware at offset 0x204 that ia10.bin or ia6b.bin does not have. ia10 and ia6b are 03000100 but FY-Q7 is 02000100
By changing FY-Q7 firmware offset 0x204 to 03000100 i thought this might be the difference between binding and not binding.
Any thoughts?

Cleric-K, 2021-09-13:
Is this the _only_ difference?

benb0jangles, 2021-09-13:
> Is this the _only_ difference?

of course there are big sections of code which are different but I am firstly searching for small differences which may be simple solutions to binding

Cleric-K, 2021-09-14:
It's quite doubtful that this would be the solution. I don't have much time now but when I get the chance I'll look at the disassembled code. Can you bind the Fly3d TX to a regular flysky rx?

benb0jangles, 2021-09-14:
> Can you bind the Fly3d TX to a regular flysky rx?

I do not have the Fly3D TX unfortunately. Only RX. If somebody has they can upload the firmware here and I will try it.